PiSugar/pisugar-power-manager-rs #6: Rustup installation failure at installing rustc component in Pi Zero W

FGMGIT, 2020-05-20:
installing component 'rustc'
31.1 MiB / 49.3 MiB ( 63 %) 1.2 MiB/s in 38s ETA: 14smemory allocation of 146176856 bytes failedAborted

SOLUTION:
rust-lang/rustup#2128

fengyc, 2020-05-21:
@FGMGIT It takes a lot of time to compile the project in Pi Zero W. For that reason, compiling in Pi zero is not recommended. I had documented the cross compiling steps in README, please feel free to open an issue if you are stuck with any problem.

FGMGIT, 2020-05-24:
Dear @fengyc can you give some help, I'm stucked with the rust installation for PiSugar 2. No way from PiZero W nor from cross compiling on macOS Catalina:

PiZero:

[image]

macOS:

[image]

fengyc, 2020-05-25:
@FGMGIT You need to clone the git project and change the current working directory to the project root directory.

FGMGIT, 2020-05-26:
**warning: spurious network error**
I'm not under proxi, firewall, etc.

[image]

fengyc, 2020-05-27:
Comment out or remove this line https://github.com/PiSugar/pisugar-power-manager-rs/blob/f1b1c11d18adb23b5a2282b479f0ed7d13201dae/.cargo/config#L3

FGMGIT, 2020-06-02:
**Build fails compiling pisugar-poweroff error: linker `arm-linux-musleabihf-ld` not found**

[image]

fengyc, 2020-06-02:
The cargo build default target has been changed to `arm-unknown-linux-musleabihf`.  If you want to compile the source code in rapi, remove the following lines: https://github.com/PiSugar/pisugar-power-manager-rs/blob/76dbc4769e6d2abe8979dbcec54c4171e68a0fd8/.cargo/config#L8-L9

richardanaya, 2020-06-20:
I was able to install 1.44.1 using RUSTUP_UNPACK_RAM=200000000 fore rustup install

fengyc, 2020-06-23:
There is no more updates recently, I will close the issue soon. Feel free to reopen it if there are more questions.

closed 2020-06-23 by fengyc