PiSugar/pisugar-power-manager-rs #7: PiZero W reboot loop when load pisugar-poweroff.service

FGMGIT, 2020-05-24:
PiZero W reboot loop when load pisugar-poweroff.service on boot process. How can I stop that service if i can't boot the PiZero?

fengyc, 2020-05-25:
The pisugar-poweroff service runs once only before poweroff. I could not reproduce the error, please provide detailed steps to reproduce. 

And please connect pizero to a monitor and see what's happening, might be something else blocks the boot procedure.

FGMGIT, 2020-05-26:
Yes it was when the system load pisugar-poweroff.service and reboot again and again. Solved powering off manually from the switch.

fengyc, 2020-05-27:
@FGMGIT Is the issue still exists in the latest build? Download the latest build from url: https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-poweroff_1.4.2_armhf.deb

fengyc, 2020-06-23:
There is no more updates recently, I will close the issue soon. Feel free to reopen it if there are more questions.

closed 2020-06-23 by fengyc