PiSugar/pisugar-power-manager-rs #8: HTTPS support

rileyinman, 2020-06-19:
I'm unsure how difficult a feature this would be to implement, but it would be nice to have the option of using HTTPS. I reverse proxy all my network services in order to use my domain's SSL certificate, but can't do that for the pisugar interface because it loads an insecure websocket.

fengyc, 2020-06-22:
A good use case. The ws url is assembled in frontend, see: 

https://github.com/PiSugar/pisugar-power-manager-rs/blob/76dbc4769e6d2abe8979dbcec54c4171e68a0fd8/electron/src/renderer/main.js#L15-L16

Change `ws` to `wss`，then build the web UI, and replace the default web UI in `/usr/share/pisugar-server/web`. Add a reverse proxy rule, nginx(or else) will set up a tunnel between wss request and pisugar backend server.

It might be better to add a http api feature, I suppose. It is not easy to setup a reverse proxy with current stream api.

rileyinman, 2020-06-22:
Unfortunately I'm not sure it works that easily, as it's giving me errors about the websockets not having a valid SSL certificate. I think some services get around this by having a self-signed certificate they distribute, but I'm not sure.

fengyc, 2020-06-23:
The pisugar server has no self-signed certificate. Could you debug the ws connection with postwoman https://postwoman.io/realtime ?

rileyinman, 2020-06-23:
Here's what my nginx rule looks like:

```
server {
        listen 0.0.0.0:80 ;
        listen [::]:80 ;
        server_name pisugar.domain ;
        location /.well-known/acme-challenge {
                root /var/lib/acme/acme-challenge;
                auth_basic off;
        }
        location / {
                return 301 https://$host$request_uri;
        }
}
server {
        listen 0.0.0.0:443 ssl http2 ;
        listen [::]:443 ssl http2 ;
        server_name pisugar.domain ;
        location /.well-known/acme-challenge {
                root /var/lib/acme/acme-challenge;
                auth_basic off;
        }
        ssl_certificate /var/lib/acme/domain/fullchain.pem;
        ssl_certificate_key /var/lib/acme/domain/key.pem;
        ssl_trusted_certificate /var/lib/acme/domain/fullchain.pem;
        location / {
                proxy_pass http://host.domain:8421;
                proxy_set_header Host $host;
                proxy_set_header X-Real-IP $remote_addr;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header X-Forwarded-Proto $scheme;
                proxy_set_header X-Forwarded-Host $host;
                proxy_set_header X-Forwarded-Server $host;
                proxy_set_header Accept-Encoding "";
        }
}

server {
        listen 0.0.0.0:8422 ;
        listen [::]:8422 ;
        server_name pisugar.rilight.tech ;
        location /.well-known/acme-challenge {
                root /var/lib/acme/acme-challenge;
                auth_basic off;
        }
        location / {
                return 301 https://$host$request_uri;
        }
}
server {
        server_name pisugar.domain ;
        location /.well-known/acme-challenge {
                root /var/lib/acme/acme-challenge;
                auth_basic off;
        }
        ssl_certificate /var/lib/acme/domain/fullchain.pem;
        ssl_certificate_key /var/lib/acme/domain/key.pem;
        ssl_trusted_certificate /var/lib/acme/domain/fullchain.pem;
        location / {
                proxy_pass http://host.domain:8422;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection $connection_upgrade;
                proxy_set_header Host $host;
                proxy_set_header X-Real-IP $remote_addr;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header X-Forwarded-Proto $scheme;
                proxy_set_header X-Forwarded-Host $host;
                proxy_set_header X-Forwarded-Server $host;
                proxy_set_header Accept-Encoding "";
        }
}
```

The debug console gives me an error of `ERR_SSL_PROTOCOL_ERROR` attempting to connect to the console, and https://postwoman.io/realtime simply gives:
```
@ 10:51:31 PM	ℹ️ [INFO]:	An error has occurred.
@ 10:51:31 PM	ℹ️ [INFO]:	
@ 10:51:32 PM	ℹ️ [INFO]:	Disconnected from wss://pisugar.domain:8422
```

fengyc, 2020-06-24:
SSL configuration is missing in the 3rd `server` rule, fix it and give a try.
```
server {
        listen 0.0.0.0:8422 ;
        listen [::]:8422 ;
        server_name pisugar.domain ;
        <ssl...>
        <ssl...>
        location / {
                proxy_pass http://host.domain:8422;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection $connection_upgrade;
                proxy_set_header Host $host;
                proxy_set_header X-Real-IP $remote_addr;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_set_header X-Forwarded-Proto $scheme;
                proxy_set_header X-Forwarded-Host $host;
                proxy_set_header X-Forwarded-Server $host;
                proxy_set_header Accept-Encoding "";
        }
}
```

rileyinman, 2020-06-24:
I added the ssl configuration, and still get `ERR_SSL_PROTOCOL_ERROR` from the wss connection. The postwoman.io output is the same as before.

fengyc, 2020-06-25:
Try the latest build https://github.com/PiSugar/pisugar-power-manager-rs/releases/tag/latest .  We don't need a standalone websocket port anymore, it has been integrated into http (`/ws`).  This should fix the problem.

rileyinman, 2020-06-27:
@fengyc Thanks! Sorry it took me a bit to get back to you. Unfortunately, it looks like the new build has some issues.
- For some reason accessing the UI directly (https://hostname:8421) using Firefox doesn't work, but it does with Chrome. Firefox gives the error `Firefox can’t establish a connection to the server at ws://hostname:8421/ws.`
- Loading the (insecure) reverse proxy on either browser fails. The log gives this error: `WebSocket connection to 'ws://pisugar.domain/ws' failed: Error during WebSocket handshake: Unexpected response code: 400`.
- Loading the secure reverse proxy fails with the same error, but for `wss://pisugar.domain/ws` instead.

fengyc, 2020-06-27:
Yeah, that is a bug, see https://github.com/1tgr/rust-websocket-lite/issues/85 . Need to wait for their updates.

rileyinman, 2020-06-27:
Ah, good to know. What about the reverse proxy failing though? It doesn't work in either browser.

fengyc, 2020-06-27:
Try this

```
server {
    <...>
    <...>
    location / {
        <...>
    }
    location /ws {
        proxy_redirect off;
        proxy_http_version 1.1;
        proxy_pass http://<host>:8421;   # Same host and port as http
        proxy_set_header Host $http_host;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
    }
}
```

rileyinman, 2020-06-27:
That worked perfectly, thank you! It even gets around the rust-websocket-lite bug :slightly_smiling_face:

rileyinman, 2020-06-27:
Should I leave this open until that bug is closed though? Since it breaks direct access this is probably something that should wait for a ralease like you said.

fengyc, 2020-06-27:
Leave this open. We will keep track of the websocket bug.

commit 7e6503e5 (2020-06-28): Fix #8
closed 2020-06-28 by fengyc