PiSugar/pisugar-power-manager-rs #9: RTC should be set to UTC time, display in either UTC or adjusted local time

rileyinman, 2020-06-24:
The problem:
- RTC clocks should be set to UTC time on Linux systems (https://tldp.org/HOWTO/Clock-2.html). Currently, the web interface sets the RTC value to local time, which breaks things.
- Not only that, it sets the clock to write-protected afterwards, which caused me [some confusion](https://github.com/PiSugar/PiSugar/issues/11) (should I open a separate issue for allowing the user to enable or disable write protection from the UI?)
- After manually setting the write protection bits to off with `i2cset` and updating the clock to UTC time, I notice that the web interface displays the UTC value but has my timezone offset labeled next to it. This is confusing.

Ideally, I believe it should function like this:
- Syncing the time to the RTC should set UTC. The user should configure their timezone as part of the system setup, which will handle DST or any other mechanisms.
- The local time should be displayed on the web interface. Perhaps there could also be an option to view the set UTC time (raw value of the RTC).

fengyc, 2020-06-25:
Thanks for the suggestions that make sense. Will fix it soon.

fengyc, 2020-06-25:
According to the RTC chip datasheet, we need to enable write protect to avoid data corruption. There is an api to sync the time from rpi to RTC `rtc_pi2rtc`.

rileyinman, 2020-06-27:
Now that the HTTPS bug is fixed, how would you suggest I test these changes?

fengyc, 2020-06-28:
The RTC clock has been set to UTC, changing system timezone does not affect RTC clock data.  If the browser timezone hadn't been changed, the RTC clock result in web UI shoud be the same as before changing system timezone.

```
system UTC 8:00   -> RTC UTC 8:00 -> Web GMT+8  16:00
system GMT+1 9:00 -> RTC UTC 8:00 -> Web GMT+8 16:00
```

fengyc, 2020-07-04:
@rileyinman The latest development build has fix the RTC timezone issue, download and have a try if you are interested in it. We will release the next version soon.

rileyinman, 2020-07-04:
Thanks, I'll test it out later today!

rileyinman, 2020-07-05:
The timezone displays correctly, and having the RTC and SYS time show separately looks great! I had some trouble getting the Web > Pi & RTC function to work, though using Pi > RTC worked fine. Also, every 3-5 seconds the "charging" indicator pops up briefly even though I'm not charging (not sure if this is related).

fengyc, 2020-07-05:
Yeah, it's confusing. Due to hardware limitation, battery charging status is dectected by reading the ouput voltage in PiSugar 2 with 4-LEDs. The voltage is not stable, especially when the power cable is plugged in. 

The good news is that has been fixed in the new models, PiSugar 2 with 2-LEDs and PiSugar 2  Pro with no-LEDs.

rileyinman, 2020-07-05:
Oh? I just got my PiSugar 2 a few days before I opened this issue, where could I find this new model?

fengyc, 2020-07-06:
Well, a long story, but we finally make it. If you feel interested, contact pisugar.zero@gmail.com for more details, we could offer you some pictures, descriptions, etc.

closed 2020-08-03 by fengyc