PiSugar/pisugar-power-manager-rs #13: pisugar-server failed to launch

fengyc, 2020-08-03:
Keep track of https://github.com/PiSugar/PiSugar/issues/12

jzting, 2020-08-03:
```
pi@OpenframeRaspi:~ $ cat /lib/systemd/system/pisugar-server.service
[Unit]
Description=pisugar-server

[Service]
EnvironmentFile=/etc/default/pisugar-server
ExecStart=/usr/bin/pisugar-server $OPTS
ExecStop=/bin/kill $MAINPID
ExecStopPost=/bin/rm -f /tmp/pisugar-server.sock
Type=simple
KillMode=process
Restart=on-failure
RestartSec=10s
WorkingDirectory=/etc/pisugar-server

[Install]
WantedBy=multi-user.target
```

fengyc, 2020-08-03:
I2C disabled? If this is a new pi, you need to enable i2c interface, see https://github.com/PiSugar/pisugar-power-manager-rs#enable-i2c-interface

fengyc, 2020-08-03:
@jzting Have you execute the `Pisugar-power-manager.sh`，or just use `dpkg -i` to install the deb file ?

jzting, 2020-08-03:
i2c was enabled, and I ran:

```
wget https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.4_armhf.deb
sudo dpkg -i pisugar-server_1.4.4_armhf.deb
```

fengyc, 2020-08-04:
Ok, a tough bug.  What's the output of `ls -al /dev/i2c*`, could you see `i2c-0` or `i2c-1`?

jzting, 2020-08-05:
I don't have the PiSugar2 hooked up right now, but this is the output:

```
pi@OpenframeRaspi:~ $ ls -al /dev/i2c*
crw-rw---- 1 root i2c 89, 3 Jul 30 16:17 /dev/i2c-3
```

The PiSugar2 provides power, but it causes [this issue](https://github.com/PiSugar/PiSugar/issues/13) when used with the HyperPixel display.

fengyc, 2020-08-07:
I'm not familiar with HypberPixel display. An I2C address collision might cause that issue, I guess. 

Pisugar server has the ability of recovering from errors, after the hardware problem is gone. 

I will keep an eye on this. Feel free to ping me if you have an update.

JdaieLin, 2026-10-01:
The reported system exposed only /dev/i2c-3. PiSugar Server now supports selecting a custom I2C bus through the i2c_bus configuration field, so it is no longer restricted to i2c-0 or i2c-1. Set i2c_bus to 3 for this configuration. Closing as resolved.

closed 2026-10-01 by JdaieLin