PiSugar/pisugar-power-manager-rs #14: seems related to #13 but that was fixed because can run without HW?

hstubbs3, 2020-09-04:
This is using a raspberry pi 4 with the hyperpixel 4.0 touchscreen ( the wide one.. - https://shop.pimoroni.com/products/hyperpixel-4?variant=12569485443155 ) 

I had followed directions on the main github page and it installed 1.4.4 today, which refused to start saying it could not load configuration. Saw there was 1.4.5 release, installed that. 

Now I can get the webUI but it does not show battery percentage. Trying the commands to the API socket, I get 'invalid request' from everything that would need to reach the hardware , such as 'get battery'.

syslog has error messages - ERROR pisugar_server] Invalid slave address : 117 

ls /dev/i2c* shows I have only a /dev/i2c-11 

The hyperpixel connects to the Pi and takes over most of the GPIO pins, but it does break out i2c on  its board. I am not familiar with this level of hardware at all, is why I bought the Pi was to learn about it and play retro arcade games, but would be more than happy to get you any logs or test something for you, just let me know what you might need.

hstubbs3, 2020-09-04:
https://github.com/pimoroni/hyperpixel

Important note
HyperPixel uses DPI mode 6, which means you can't use (hardware) I2C or SPI at the same time (the setup.sh script will disable those interfaces for you, but make sure not to reenable them by accident).

In addition, DAC type of products communicating with the Pi over I2S are also incompatible, as they use the same pins. It is possible to use the on-board audio chip alongside HyperPixel however, provided you force route the audio signal over HDMI, or are happy losing refined control over the backlight (PWM).

d'oh...

fengyc, 2020-09-05:
Thanks for the information. 

We had read the note carefully, but unfortunately there is no solution for now.  As the I2C funstion has been broken by HyperPixel, we could do nothing in software level to fix the problem.

hstubbs3, 2020-09-07:
Exactly, it is not a software issue. 

I am closing this as I have figured out the workaround - I blocked the pogo pins for SDA and SCL with electrical tape and soldered wires to the SDA and SCL pad and connected those to the HyperPixel's I2C bus. 

The HyperPixel I2C showed as /dev/i2c-7...  I created a symbolic link /dev/i2c-1 => /dev/i2c-7 and your software was able to connect and read the battery life just fine.

closed 2020-09-07 by hstubbs3