PiSugar/pisugar-power-manager-rs #18: auto_power_on

3dlabsio, 2020-12-27:
Does this new feature allow us to not have to press the "activate" button if the battery is disconnected or loses all charge?

fengyc, 2020-12-28:
No, it didn't. "activate" button and `auto_power_on` are different funtions.

In order to prevent the battery from being damaged,  the chip (on PiSugar2 Pro) will stop the power output when the battery loses all charge. We need to press the "activate" button to reenable it.

3dlabsio, 2020-12-28:
Is there a connector on the board to place the activate button outside of a case?

cariacou, 2021-11-23:
I am having the same issue. The Pi is in a box, with no access to the "activate" button. 

Is there any way to use the PI Sugar, without using the buttons, by simply just plugging in / unplugging the USB-C cable?

nzola, 2024-11-05:
Pisugar when plugged on power adapter turns on, but pressing power buttons on Pisugar board does nothing. It was working before. I could turn it on and off without any problems, because I know the on and off sequences.  
After the battery is fully charged, I unplug the power adaptor from Pisugar, it still on for many hours with the supply from the battery. The on and off switch still doesn't work.

JdaieLin, 2026-10-02:
This behavior is a hardware safety limitation rather than a software defect. After the battery has been disconnected or completely discharged, the protection circuit requires the physical Activate button to re-enable battery output. Automatic power-on cannot bypass that protection state. The original question has been answered, so this issue is being closed.

closed 2026-10-02 by JdaieLin