PiSugar/pisugar-power-manager-rs #21: How to prevent from logging to syslog ?

Cocoo68332, 2021-01-08:
Hi,

I am monitoring the battery by rpimonitor which sends get* to the server pisugar-server each 10sec.

pisugar-server logs each access to /var/log/syslog,  saturating the log.

I tried preventing the server from loging to syslog, but cant find an option. My server is launched with the default options (from /etc/default/pisugar-server) :
  
OPTS=--config config.json --model 'PiSugar 2 Pro' --web /usr/share/pisugar-server/web --http 0.0.0.0:8421 --ws 0.0.0.0:8422  --tcp 0.0.0.0:8423 --uds /tmp/pisugar-server.sock

I tried the -s option, but it does not change anything.

Is it possible to add an argument to -s to give a specific log file ? I can then pit /dev/null, or add an option to prevent from logging each access to the server ? or the ones from localhost ?

Thx !

L

 Jan  8 15:31:55 pi4 pisugar-server[341]: [2021-01-08T14:31:55Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46092
Jan  8 15:31:55 pi4 pisugar-server[341]: [2021-01-08T14:31:55Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46094
Jan  8 15:31:55 pi4 pisugar-server[341]: [2021-01-08T14:31:55Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46096
Jan  8 15:32:05 pi4 pisugar-server[341]: [2021-01-08T14:32:05Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46098
Jan  8 15:32:05 pi4 pisugar-server[341]: [2021-01-08T14:32:05Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46100
Jan  8 15:32:05 pi4 pisugar-server[341]: [2021-01-08T14:32:05Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46102
Jan  8 15:32:05 pi4 pisugar-server[341]: [2021-01-08T14:32:05Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46104
Jan  8 15:32:06 pi4 pisugar-server[341]: [2021-01-08T14:32:06Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46106
Jan  8 15:32:06 pi4 pisugar-server[341]: [2021-01-08T14:32:06Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46108
Jan  8 15:32:06 pi4 pisugar-server[341]: [2021-01-08T14:32:06Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46110
Jan  8 15:32:06 pi4 pisugar-server[341]: [2021-01-08T14:32:06Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46112
Jan  8 15:32:06 pi4 pisugar-server[341]: [2021-01-08T14:32:06Z INFO  pisugar_server] Incoming tcp connection from: 127.0.0.1:46114

fengyc, 2021-01-10:
If `-s` or `--syslog` is not specified, pisugar-server would not write syslog. These logs are generated by systemd. The pisugar-server is managed by systemd, and all the log is captured by systemd-journald.

Maybe you need to add a journald configuration to disable `ForwardToSyslog`, see `man systemd` and `man journald`.

Cocoo68332, 2021-01-10:
Thx for the advice. I was able to stop the log by adding to /etc/systemd/journald.conf : 

MaxLevelSyslog=err

However, this is not a good solution because 1) it prevents ALL units logs, and 2) I am affraid to miss an alert message in case of power failure ?

To solve pb 1) I tried to add to /usr/lib/systemd/system/pisugar-server.service

LogLevelMax=err

But for a reason that I dont understand, this does not work, the log goes on.

 So I backtracked to the forst solution to alter journald.conf.

fengyc, 2021-02-04:
See https://www.freedesktop.org/software/systemd/man/systemd.exec.html#StandardOutput=
```
StandardOutput=append:/tmp/pisugar.log
StandardError=append:/tmp/pisugar-error.log
```

closed 2021-02-21 by fengyc