PiSugar/pisugar-power-manager-rs #22: How to force RTC update on boot?

GSVNoFixedabode, 2021-02-03:
Is there a simple command to immediately update the date on a Pi Zero from the Sugar2 RTC on boot? 
At present I have a script that runs on boot (within crontab, @reboot...):
> - take a camera snapshot using datetime as name; 
> - sends to FTP server;
> - resets next wake time, either next hour or next day; 
> - shutdown.
But it seems when Pi Zero first starts ,the internal clock is still set to time at last shutdown so files are being mis-named. Is there a way to add a step to immediately update date from the RTC? Date seems to auto update from board after a few minutes, but need to do it earlier.  Do I need to remove fake-hwclock app from Pi as you do when adding an external RTC board?

fengyc, 2021-02-04:
Add the following code into the `@reboot` script
```
echo 'rtc_rtc2pi' | nc -q 0 127.0.0.1 8423
```
See https://github.com/PiSugar/pisugar-power-manager-rs#unix-domain-socket--webscoket--tcp

GSVNoFixedabode, 2021-02-04:
Ah, thank you - I'd missed that when looking for the command!

closed 2021-02-04 by GSVNoFixedabode