PiSugar/pisugar-power-manager-rs #23: Cant install .debs

PeterMalkin, 2021-03-27:
I am generating an sdcard image for my project using QEMU.
(https://wiki.debian.org/RaspberryPi/qemu-user-static)

It seems your deb packages fail to install

+ mktemp
+ TEMP_DEB=/tmp/tmp.kpff6YLkzb.deb
+ wget -O /tmp/tmp.kpff6YLkzb.deb http://cdn.pisugar.com/release/pisugar-poweroff_1.4.9_armhf.deb
--2021-03-27 22:03:23--  http://cdn.pisugar.com/release/pisugar-poweroff_1.4.9_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.22.184
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.22.184|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 276934 (270K) [application/x-debian-package]
Saving to: ‘/tmp/tmp.kpff6YLkzb.deb’

/tmp/tmp.kpff6YLkzb.deb   100%[====================================>] 270.44K   672KB/s    in 0.4s    

2021-03-27 22:03:24 (672 KB/s) - ‘/tmp/tmp.kpff6YLkzb.deb’ saved [276934/276934]

+ dpkg -i /tmp/tmp.kpff6YLkzb.deb
Selecting previously unselected package pisugar-poweroff.
(Reading database ... 95882 files and directories currently installed.)
Preparing to unpack /tmp/tmp.kpff6YLkzb.deb ...
Running in chroot, ignoring request: daemon-reload
Unpacking pisugar-poweroff (1.4.9) ...
Setting up pisugar-poweroff (1.4.9) ...
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
Created symlink /etc/systemd/system/shutdown.target.wants/pisugar-poweroff.service → /lib/systemd/system/pisugar-poweroff.service.
+ rm -f /tmp/tmp.kpff6YLkzb.deb

+ mktemp
+ TEMP_DEB=/tmp/tmp.kpff6YLkzb.deb
+ wget -O /tmp/tmp.kpff6YLkzb.deb http://cdn.pisugar.com/release/pisugar-poweroff_1.4.9_armhf.deb
--2021-03-27 22:03:23--  http://cdn.pisugar.com/release/pisugar-poweroff_1.4.9_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 47.246.22.184
Connecting to cdn.pisugar.com (cdn.pisugar.com)|47.246.22.184|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 276934 (270K) [application/x-debian-package]
Saving to: ‘/tmp/tmp.kpff6YLkzb.deb’

/tmp/tmp.kpff6YLkzb.deb   100%[====================================>] 270.44K   672KB/s    in 0.4s    

2021-03-27 22:03:24 (672 KB/s) - ‘/tmp/tmp.kpff6YLkzb.deb’ saved [276934/276934]

+ dpkg -i /tmp/tmp.kpff6YLkzb.deb
Selecting previously unselected package pisugar-poweroff.
(Reading database ... 95882 files and directories currently installed.)
Preparing to unpack /tmp/tmp.kpff6YLkzb.deb ...
Running in chroot, ignoring request: daemon-reload
Unpacking pisugar-poweroff (1.4.9) ...
Setting up pisugar-poweroff (1.4.9) ...
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
Created symlink /etc/systemd/system/shutdown.target.wants/pisugar-poweroff.service → /lib/systemd/system/pisugar-poweroff.service.
+ rm -f /tmp/tmp.kpff6YLkzb.deb

fengyc, 2021-03-28:
The postinst script (https://github.com/PiSugar/pisugar-power-manager-rs/blob/master/pisugar-server/debian/postinst) would start the systemd service after the deb package has been installed.  

I had fix the problem. Please download and test the latest development build (https://github.com/PiSugar/pisugar-power-manager-rs/releases/tag/latest)

PeterMalkin, 2021-03-28:
Thanks for your attention! 

Although your fix is a good thing to have in the postinst script, the problem I believe lies in a different place.

++ mktemp
+ TEMP_DEB=/tmp/tmp.CKTygFMdII.deb
+ wget -O /tmp/tmp.CKTygFMdII.deb https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.9_armhf.deb
--2021-03-28 19:18:25--  https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-server_1.4.9_armhf.deb

HTTP request sent, awaiting response... 200 OK
Length: 1306406 (1.2M) [application/octet-stream]
Saving to: ‘/tmp/tmp.CKTygFMdII.deb’
/tmp/tmp.CKTygFMdII.deb   100%[====================================>]   1.25M  2.51MB/s    in 0.5s    

2021-03-28 19:18:26 (2.51 MB/s) - ‘/tmp/tmp.CKTygFMdII.deb’ saved [1306406/1306406]

+ dpkg -i /tmp/tmp.CKTygFMdII.deb
Selecting previously unselected package pisugar-server.
(Reading database ... 95866 files and directories currently installed.)
Preparing to unpack /tmp/tmp.CKTygFMdII.deb ...
Running in chroot, ignoring request: daemon-reload
Unpacking pisugar-server (1.4.9) ...
Setting up pisugar-server (1.4.9) ...
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
Created symlink /etc/systemd/system/multi-user.target.wants/pisugar-server.service → /lib/systemd/system/pisugar-server.service.
Running in chroot, ignoring request: is-active
Running in chroot, ignoring request: start
+ rm -f /tmp/tmp.CKTygFMdII.deb
++ mktemp
+ TEMP_DEB=/tmp/tmp.Muh1tSNGXg.deb
+ wget -O /tmp/tmp.Muh1tSNGXg.deb https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-poweroff_1.4.9_armhf.deb
--2021-03-28 19:18:29--  https://github.com/PiSugar/pisugar-power-manager-rs/releases/download/latest/pisugar-poweroff_1.4.9_armhf.deb

Saving to: ‘/tmp/tmp.Muh1tSNGXg.deb’

/tmp/tmp.Muh1tSNGXg.deb   100%[====================================>] 269.90K  1.56MB/s    in 0.2s    

2021-03-28 19:18:30 (1.56 MB/s) - ‘/tmp/tmp.Muh1tSNGXg.deb’ saved [276380/276380]

+ dpkg -i /tmp/tmp.Muh1tSNGXg.deb
Selecting previously unselected package pisugar-poweroff.
(Reading database ... 95882 files and directories currently installed.)
Preparing to unpack /tmp/tmp.Muh1tSNGXg.deb ...
Running in chroot, ignoring request: daemon-reload
Unpacking pisugar-poweroff (1.4.9) ...
Setting up pisugar-poweroff (1.4.9) ...
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
mount: /config/device-tree: mount point does not exist.
* Failed to mount configfs - 2
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.4.0-70-generic/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic
Created symlink /etc/systemd/system/shutdown.target.wants/pisugar-poweroff.service → /lib/systemd/system/pisugar-poweroff.service.
+ rm -f /tmp/tmp.Muh1tSNGXg.deb

I think one of those errors is resolved by installing libi2c-dev before installing the debian packages for pisugar.
Are there any other packages that need to be installed prior to installing your .debs?

fengyc, 2021-03-30:
The log said "Module i2c-dev not found in directory /lib/modules/5.4.0-70-generic".  But the search result of libi2c-dev said that it is a userspace programming library, see https://packages.ubuntu.com/groovy/libi2c-dev. So, did the i2c-dev module had been configured correctly ?

JdaieLin, 2026-10-02:
This report concerns the 1.4.9 Debian packages and an old QEMU/chroot installation flow. The package post-install scripts and release packaging have changed substantially since then, so the original logs no longer describe the current packages. Closing as obsolete. If the current release still fails in a chroot, please open a new issue with the current version, target image, and complete dpkg output.

closed 2026-10-02 by JdaieLin