PiSugar/pisugar-power-manager-rs #25: PiSugar Power Management page not showing battery levels

causalloop, 2021-05-03:
Checked on firefox and chrome as well as windows and os x to be sure, but I just see "...%" on the manager page.  

[image]

I'm 95% that I have the 2 LED version (I'm assuming thats referring to the 2 blue led's on either side of the board - the amazon page wasn't very specific).  Its definitely the 1200mAh battery for the Pi Zero.  I've got some questions about cmd line usage but I'll save those for later.

causalloop, 2021-05-04:
Just to add to this, executing:
`echo "get battery" | nc -q 0 127.0.0.1 8423` returns "Invalid request"

wlmh110, 2021-05-07:
Hi，
Could you please use i2c tools to see what is in 0x75?
i2cdump -y 1 0x75

causalloop, 2021-05-07:
Curious, the first time I ran `i2cdetect -y 1` (before I posted), it definitely showed up in the table, but now it doesn't.  The i2cdump returns: 
```i2cdump -y 1 0x75
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
10: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
```

causalloop, 2021-05-07:
Just tried it on another pi zero w and the results were the same.  Sounds like something's up with the board.

aantuness, 2021-05-21:
Have the same issue, i noticed using i2cdetect -y 1 that I have 32 but not 75..

18637, 2021-06-20:
I also have the same issue. mine is running i2c 32 also. Running on a Rp4 Compute module. is there any way to fix this?

fengyc, 2021-06-26:
Steps to figure out what went wrong:
1. Ensure I2C is enabled `sudo raspi-config`
2. Remove PiSugar, then run `i2cdetect -y 1 0x75`. If there is any output, it means another device with same I2C address 0x75 conflicts with PiSugar.  You have to remove/disable the I2C pins of either that device or PiSugar, otherwise both of them do work.

18637, 2021-06-26:
> Steps to figure out what went wrong:
>
> 1. Ensure I2C is enabled `sudo raspi-config`
> 2. Remove PiSugar, then run `i2cdetect -y 1 0x75`. If there is any output, it means another device with same I2C [...]

@fengyc  Are you saying that if you disable the i2c pins of either device then they will both work? or wont work?
Can the Pi sugar i2c address be changed?

psi5asp, 2021-06-26:
Pi-4 requires 
/boot/config.txt 
dtparam=i2c1=on

18637, 2021-06-27:
@fengyc The latest version 1.5.0 has fixed this issue for me. Thank you.

aotd1, 2021-10-05:
> Curious, the first time I ran `i2cdetect -y 1` (before I posted), it definitely showed up in the table, but now it doesn't.

Possible very simple root case: your raspberry pi powered directory over USB and PiSugar2 is completely powered off by switch =)
Just power on PiSugar =)

alcofrisbas, 2021-12-01:
I am running pisugar-server v1.6.1 on my pi4(32bit) and am still having this problem. I have it plugged into an hdmi cable with no other hats or connections, and still only 0x32 is the only recognized i2c connection. I have followed all advice in this thread so far, but still have no percentage readout nor any connection on address 0x75. What are my next steps?

fengyc, 2021-12-02:
> I am running pisugar-server v1.6.1 on my pi4(32bit) and am still having this problem. I have it plugged into an hdmi cable with no other hats or connections, and still only 0x32 is the only [...]

Hi, @alcofrisbas please check whether the pisugar is power on or off, if it is power on and i2c address 0x75 could not be detected, contact us pisugar.zero@gmail.com.

alcofrisbas, 2021-12-02:
@fengyc I am not sure what you mean. The battery pack is the only power to the pi, so it has to be on in order for the pi to be on.

fengyc, 2021-12-02:
> @fengyc I am not sure what you mean. The battery pack is the only power to the pi, so it has to be on in order for the pi to be on.
 
@alcofrisbas  Ok, please contact us via email,  our customer support will track this case.

iamk3, 2022-04-07:
Has this issue been resolved? I'm seeing the same thing.

MrTrueFail, 2022-05-22:
Hi,
I have the same issue: No battery seems to be available, i2cdump is empty and requesting a state via the socket returns invalid request.
However, when I change Schedule Wake Up from "On Power Restore" to "disable" and back again to "On Power Restore", suddenly everything is working the way it should.
I launched the server manually to get a output and apparently the pisugar_core module is stuck initializing the battery?
[image]

Weird think is that it doesn't happen every time I boot up the pi zero. Maybe some timing issue?

wlmh110, 2022-05-29:
> Hi, I have the same issue: No battery seems to be available, i2cdump is empty and requesting a state via the socket returns invalid request. However, when I change Schedule Wake Up from "On Power [...]

Please use this command and post the result:
i2cdetect -y 1

H3CATE, 2023-12-28:
Just got a board today and I got the exact same issue. Is my board defective ? ordered it from Amazon

JdaieLin, 2026-10-01:
The missing battery data and I2C-not-connected symptoms can be caused by poor contact between the PiSugar pogo pins and the solder points on the underside of the Raspberry Pi GPIO header. The official FAQ provides the supported troubleshooting procedure and safety precautions: https://docs.pisugar.com/docs/product-wiki/battery/faq#1-why-pisugar-is-not-able-to-power-raspberry-pi-why-pisugar-23-cannot-be-detected-on-i2c-bus

Power the Raspberry Pi off first. Gently clean the underside solder points with alcohol or a plastic tool, reinstall PiSugar, and test again. If you are comfortable soldering, the FAQ also suggests adding a small amount of solder to improve contact. Please report back if the device is still missing from the I2C bus after following the FAQ.

closed 2026-11-15 by github-actions[bot]