PiSugar/pisugar-power-manager-rs #29: [BUG]

ArJay60, 2021-07-13:
PI3B with PiSugar Pro and Pisugar powermanager.

When rebooting Pi from terminal with 'reboot'-command the power to the Pi is cut off. The Pi needs to be restarted by pressing the activate button in the Pisugar.

Expected is that with a reboot command the Pi remains under power and can restart without further manual intervention.

Do i do something wrong why it is not working?

fengyc, 2021-07-15:
@wlmh110

wlmh110, 2021-07-15:
Hello, please contact us to get a replacement.
pisugar.zero@gmail.com

psi5asp, 2021-07-23:
It is possible that the unit is under powered and the battery level goes to zero (espeically with USB drives).

Sometimes Pi-3B's have underpower issues - especially under heavy load or USB/HDD usage.

I had an issue where the battery level was decreasing even though it was showing as 'Charging' on the PiSugar - I suspected it was due to power supply issues to the Pi 3B (connnected to HDMI showed power symbol occasionally). Indicative using the `vcgencmd get_throttled` command. So I changed to using a Pi-4 PSU (ie 5V 3amps) connected directly tho the PiSUgar to ensure there was enough power and seemed to fix it.

The PiSugar Pro can handle 2 types of PSU inputs - since these can also be used on Pi-4's, the Pi-4 UPS work as well.

ArJay60, 2021-07-23:
Not the issue. Dmesg gives no low voltage warnings. Batteries are loaded. Pi is running happely without issues for days. If i pull the power goed from the pisugar everything jeeps running fine. The only thing i should not do is using the reboot command on the pi to reboot the Pi. If info that everything including pisugar is powered off.

ArJay60, 2021-07-23:
Last sentence should be: If i do that everything including pisugar is powered off. Restart only possible by pressing button on pisugar.

JdaieLin, 2026-10-02:
This report is from 2021 and does not include a reproducible case on a currently supported release. Power-off, reboot, RTC, and auto-power-on handling have changed substantially since then. Closing as stale/obsolete. If reboot still removes power with the current release, please open a new issue with the PiSugar model, firmware version, server version, OS version, and service logs.

closed 2026-10-02 by JdaieLin