PiSugar/pisugar-power-manager-rs #30: I'd like for the battery management to control powering off easier.[FEATURE]

snarflakes, 2021-08-16:
**Is your feature request related to a problem? Please describe.**
users right now have to 
1. press a button my pi device to shut down the application and the raspberry pi.
2. then Wait till the green light finishes processing - 10s?
3. THEN they have to flip the pi-sugar switch to turn it off.

**Describe the solution you'd like**
I want my users to flip the switch to turn it on,
and flip the switch to turn the unit off (not have to press a button to shutdown the raspberry pi and program first). The switch would shutdown raspberrypi program and raspbian first then stop feeding power for a completely safe shutdown process.

fengyc, 2021-08-17:
We need more information, the pisugar-server and  pisugar-poweroff version. 

Generally, if you have a customized tap action shell script to shutdown the raspberry pi, then the pisugar-poweroff will shutdown the pisugar.

fengyc, 2021-08-17:
@wlmh110

snarflakes, 2021-08-17:
Thanks for the help!

So properly configuring the pisugar-poweroff software would enable my own GPIO button activation programmed to shutdown ras pi (not the button on the pisugar?) to completely cut power to the whole board (all lights turn off)?  

So in the end, the power switch would not be tied to a direction for on and off? If its off, flip the switch to the opposite direction (could end up being any direction)?  The power switch ends up only being for powering ON the device right?
thanks

fengyc, 2021-08-17:
The pisugar-poweroff needs no configuration except the pisugar model. Before the Linux os shutdown, systemd would launch pisugar-poweroff, and then it would send an i2c message to pisugar to cut off the power supply. 

So, you don't need to flip the power switch to shutdown the pisugar. But you need to flip the switch off and then on to wake up pisugar and resume the power supply.

snarflakes, 2021-08-17:
What software needs to be pre-loaded for that to automatically take place?

this one?
https://github.com/PiSugar/PiSugar/wiki/PiSugar-Power-Manager-(Software)

fengyc, 2021-08-17:
The latest release: https://github.com/PiSugar/pisugar-power-manager-rs/releases/tag/v1.5.2

If you need to customize the tap button action, you need to install pisugar-server and pisugar-poweroff.

The pisugar-server web console url is http://<your rpi ip>:8423

snarflakes, 2021-08-17:
Thanks, your team is the best. 
I'm going to test the pi-sugar power manager.

I love the idea of the easy off function of the pi-sugar power manager.  But in terms of UI for powering ON, I'm not sold on the flip switch back and forth for "wake" function.  Seems quite confusing for new users of a product to flip the switch back and forth to turn it on. That would take special very clear user instruction.  But I love that its an option.  So thanks for all your attention to detail.

fengyc, 2021-08-17:
Yeah, we had made some improvements on the 3rd generation.

snarflakes, 2021-08-17:
Last question:
The battery protection “automatic shutdown for low battery level” function is that loaded automatically just like the raspbery pi power off automated sleep state? Or is that require the full server set up?

Thanks

Sent from ProtonMail Mobile

fengyc, 2021-08-17:
Any thing that could be runnable. It could be a shell command, or a runnable file,  ".sh" or ".py" (after `chmod +x `)

snarflakes, 2021-08-17:
Where does the script need to be located for the software to find it?

fengyc, 2021-08-18:
Anywhere is fine. You need to ensure the full path is correct.

closed 2021-08-19 by snarflakes

erlendv, 2022-01-15:
Is there not any way to restart the pisugar when re-powered without flipping the on/off switch? I just received my pisugar 2 after a month in shipping just to realize this. Its a breaker for my application since it will be mounted inside an old retro speaker.

fengyc, 2022-01-17:
@erlendv Could you open another issue, and describe the purpose and the problem?  Thanks.