PiSugar/pisugar-power-manager-rs #37: Cannot process pisugar-power-manager-rs with cargo-aur

SecretShangri-La, 2021-11-13:
I wasn't sure if I should report this issue to cargo-aur, but other crates (i.e. sharkdp/bat) handled it fine, so I decided it was a problem specific to this crate.

### Is Problem
```bash
:: Error: missing field `package` at line 9 column 1
````
As it turns out, I'm getting this error and cannot generate a PKGBUILD.

I decided to use PiSuger 2 Pro to build my own computer set with Rasbarry Pi 3 to do programming exercises and research for college. I have chosen the Manjaro Gnome to run satisfactorily on the inefficient Pi 3.
However, since Manjaro is an Arch system, and not any other RPM or APT system, it is not possible to use the ( https://github.com/PiSugar/pisugar-power-manager-rs/issues/26#issuecomment-843800900 ) I took the choice. This is because a modern graphical interface is essential for the exercise.

Please let us have an ArchLinux package as soon as possible.
Thank you very much.

fengyc, 2021-11-16:
Hi, @SecretShangri-La . See https://github.com/PiSugar/pisugar-power-manager-rs#install-archlinux-only ,  I had added a  script  to generate the package for archlinux.

Please let me know whether it solve the problem or not.

SecretShangri-La, 2021-11-19:
I'm going to be straight to the point. The `tar.gz` format files, at least the ones starting with `pisugar-archlinux`, don't seem to exist to me. I've also tried `cargo aru`, but it's not practical. If I'm missing something, I'm very sorry, but are you sure that's been released?

[image]

SecretShangri-La, 2021-11-19:
When I was building DoxBox-X a while ago, I noticed that a shell script called `build-aur.sh` was added. It seems that I was mistaken. I would like to take this opportunity to apologize.

Now, let's get down to business. When I read the contents of the `build-aur.sh` script, I found that it was intended to be run inside this project, so I ran it, and the following log was output

```bash
cat: PKGBUILD: No such file or directory
tar: Removing leading `/' from member names
tar: /run/media/my_user_name/D40A-1102/target/arm-unknown-linux-musleabihf/release/pisugar-server: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-server/.rpm/_ws.json: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-server/.rpm/config.json: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-server/.rpm/pisugar-server.default: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-server/.rpm/pisugar-server.service: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/electron/dist/web: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/target/arm-unknown-linux-musleabihf/release/pisugar-poweroff: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-poweroff/.rpm/pisugar-poweroff.default: Cannot stat: No such file or directory
tar: /run/media/my_user_name/D40A-1102/pisugar-server/.rpm/pisugar-poweroff.service: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors

```

I also ran it outside the project directory (again, both root and non-root) just to be sure, but it only generated a compressed file.

SecretShangri-La, 2021-11-19:
Here's what I found out after dissecting an interesting shell script and PKGBUILD file.
As it turns out, these two scripts do not work as the current situation. They need to be hacked a bit. However, I have a strong interest in this hack, so I will continue. If there is any further news, I will post it in this thread.
Thanks for creating the script!

fengyc, 2021-12-11:
Hi, @SecretShangri-La . I had fix some errors of the script. As I'm not familiar with arch linux,  I am not sure it works or not (tested in a docker box that without systemd as init system).  If you are still working on it,  you could take a look. Thanks for your time!

JdaieLin, 2026-10-01:
Arch/AUR support for this issue was added in commit 54503ef and the packaging script received subsequent fixes. The old AUR build workflow has since been retired, so this historical issue is complete/obsolete and is being closed.

closed 2026-10-01 by JdaieLin