PiSugar/pisugar-power-manager-rs #43: [FEATURE] Is there a way to turn off the blue LED lights?

sxxymzy, 2021-12-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
 Is there a way to turn off the blue LED lights?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Need to turn off the blue LED lights

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
None

**Additional context**
Add any other context or screenshots about the feature request here.

fengyc, 2021-12-15:
Sorry, the power led is directly connected to the power switch, it could not be turn off individually.

JdaieLin, 2026-10-02:
As explained above, the blue power LED is electrically connected to the power switch and cannot be controlled independently in software. This is a hardware limitation, so the requested feature cannot be implemented. Closing as not planned.

closed 2026-10-02 by JdaieLin