PiSugar/pisugar-power-manager-rs #44: [BUG] Raspberry pi zero 2 with pisugar2

Jesstr8803, 2021-12-20:
**Environment**
 - Pi hardware version: Zero 2
 - OS version: Pi OS lite Buster
 - Application version: 1.5.0

**To Reproduce**
Steps to reproduce the behavior:
After installing 1.5.0 of piserver using curl http://cdn.pisugar.com/release/Pisugar-power-manager.sh | sudo bash
The battery seems to communicated fine with raspberry pi, can request any battery flag using nc -U /tmp/pisugar-server.sock and then typing in a request. But the RTC return nothing but Invalid request.

**Expected behavior**
Should be able to request data from RTC

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
I2C tools shows both i2c addresses.

```
pi@DropCam:~ $ sudo nc -U /tmp/pisugar-server.sock
get model
model: PiSugar 2 (2-LEDs)
get rtc_alarm_time
Invalid request.
get rtc_time
Invalid request.
get battery_i
battery_i: -0.17294818
```

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- 32 -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- 75 -- --
```

```
pi@DropCam:~ $ i2cdump -y 1 0x32
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 13 38 a1 01 20 12 21 00 13 21 7f 20 12 21 00 00    ?8?? ?!.?!? ?!..
10: 52 0b 00 00 93 79 bf bf 02 be fb eb fe df eb ff    R?..?y?????????.
20: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
30: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
40: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
50: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
```

I tried using all the different versions of PiSugar server/power manager from 1.5.0 to 1.6.3, they all gave different results, I could almost make it work with 1.6.3, but I would have to go back and manually configure /etc/defaults/pisugar-server, and /etc/pisugar-server/config.json, as the default file was messed up because of the OSD menu(not actually sure what it is called, but it's the blue screen) that asks about the type of PiSugar, and the login for the config.json was messed up, the user would say OK, I'm thinking its becasue of the OSD menu. But when I did this I could communicate with the RTC, but the behaviour wasn't right either. When setting the alarm the time would populate correctly but the date would always have the year 2020 or 1999. I also tried installing 1.6.3 then down grading to 1.5, and it would work better the dates would still be messed up. 

Not sure what is causing the breakdown, In the meantime ive gone back to using a Pi Zero. I would love to be able to use my Pi Zero 2 as it is much faster, although a little more power hungry. But for my application which is a long duration timelapse, I need to be able to boot fast take a picture and and shutdown fast. then wait until the next photo to start the cycle again.

fengyc, 2021-12-22:
@Jesstr8803 Thanks for the report. 

If the  pisugar2 rtc keep reporting  lots of `xx`s, that  means the hardware has some problem, please contact pisugar.zero<at>gmail.com for further support.

The error in ` /etc/default/pisugar-server` had been fixed in the latest development build, you could install the latest one and test again.

Jesstr8803, 2021-12-26:
@fengyc Thanks for getting back to me, I was able to try 1.6.4 and that version seemed to fix the /etc/default/pisugar-server and the web authorization login. But I am still having issues with the date reporting back correctly with the rtc_alarms. I have contacted the email provided. Thank you fengyc. I also have a couple more boards on order, ,so when I recieve them I can do more testing!

Jesstr8803, 2021-12-28:
@fengyc I have contacted the email provided, and they said that uaving thr XX in the RTC is ok and it should not affect the function of the RTC as the important information is at the top of the register. Looking forward to the next update, hopefully you can figure out the date issue with the alarms!

fengyc, 2021-12-29:
Could you update the rtc and set up a wake up alarm via web ui ?

`rtc_alarm_time: 1999-12-31T17:00:00-07:00`  is `UTC 2000-01-01T00:00:00`,  the pisugar-server would return this value if the data of rtc is incorrect.

Jesstr8803, 2021-12-31:
@fengyc, I can set the time correctly from both the Web UI and using the socket. But the date still comes back incorrectly.

```
pi@DropCam:~ $  sudo nc -U /tmp/pisugar-server.sock
get rtc_alarm_time
rtc_alarm_time: 2000-01-01T13:47:00-07:00
rtc_alarm_set 2021-12-31T13:48:00-07:00 127
rtc_alarm_set: done
get rtc_alarm_time
rtc_alarm_time: 2000-01-01T13:48:00-07:00
```

fengyc, 2022-01-04:
@Jesstr8803 The rtc_alarm_time value is correct. Only the hh:MM:ss and the weekday repeat values could control the rtc alarm, the yyyy/mm/dd part of rtc_alarm_time would be ignored.

Jesstr8803, 2022-01-15:
@fengyc, Sorry for taking so long to get back, Thanks for all your help! Just out of curiosity, did the date report correctly in a previous version? maybe 1.5?

fengyc, 2022-01-17:
@Jesstr8803 No.  

My plan is to add a new feature to allow setting the year and month of the alarm datetime, but I am not sure whether the RTC chip could support this feature. Will update the code if I make it work.

JdaieLin, 2026-10-02:
The maintainer clarified that this RTC only uses the alarm time and weekday-repeat fields; the displayed year/month/day portion is not stored as a full calendar alarm and is therefore ignored. The remaining installation details refer to old 1.5/1.6 packages and configuration scripts that have since been replaced. Closing as answered/obsolete.

closed 2026-10-02 by JdaieLin