PiSugar/pisugar-power-manager-rs #61: PiSugarSButtonActive.sh and safe shutdown on S plus?

RedStonesparrow, 2022-04-13:
having trouble figuring out what i need to download for my S plus to shutdown the pi when the power is low or when I pres the button. I saw the  PiSugarSButtonActive.sh but i'm unsure what it does and how to use it properly.

g-or-well, 2022-07-07:
For the "button shutdown" on my PiSugar S (without the plus) my approach was this:

- get said script and edit line 37 and/or 39 to read "sudo shutdown -h now". 
- edit your crontab to start the script at boot
- push the button: magic :-)

As for the automatic shutdown on low battery: isn't that a feature anyway? 
In the wiki it says: "When the battery voltage is lower than 3V, PiSugar S will shut down immediately". Mine never ran empty, but I would read "shut down" as a regular shutdown like you could do with "shutdown -h now".
Otherwise you would need to read the battery level and react to that. No idea how to do that but it seems PiSugar power manager might be able to accomplish that.

midicdj1000, 2023-08-02:
I know very little about linux so  how do I get the script, how do I edit the script, how do I edit the crontab ?

JdaieLin, 2026-10-02:
This is a usage/support question rather than a defect in the Rust power manager. Current PiSugar S documentation and example scripts are available here:

https://docs.pisugar.com/docs/product-wiki/battery/pisugar-s-series
https://docs.pisugar.com/docs/product-wiki/battery/pisugar-scripts-example

Closing as answered. If a current documented script fails, please open a new issue with the exact PiSugar S model, OS, script, and error output.

closed 2026-10-02 by JdaieLin