PiSugar/pisugar-power-manager-rs #74: Pisugar 3 RTC Model and Configuration

gianmarco-mameli, 2022-08-04:
Hi,
i've recently buyed a Pisugar 3, for using it with a Raspberry PI Zero W and the Pwnagotchi image.
I need to use the hardware RTC onboard, but unfortunately there's no way to make it work. Before buiyng the Pisugar, the Raspberry was equipped with a hardware rtc from Adafruit and it worked like a charm. I uninstalled it before installing the Pisugar, and started with a fresh os image.
With the i2cdetect -y 1 command I correctly see the 0x68 address, but I don't find any overlay or config to make it work. I've also tried this guide whit no luck [https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time](https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time)

Anyone has some info or solution, or knows the precise model of the RTC on board of Pisugar 3?

Thanks

mitpre, 2022-08-14:
I have the same issue and the creators are unresponsive over their email.

gianmarco-mameli, 2022-08-17:
ok, @mitpre thanks for sharing

mitpre, 2022-09-02:
@gianmarco-mameli I got a new update from the team behind pisugar3. You can now use on board RTC. Guide on how to do so is here: https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series#rtc-on-board

gianmarco-mameli, 2022-09-02:
Hi, @mitpre thank you, I'd give it a try asap and I write back to you to confirm

gianmarco-mameli, 2022-10-31:
Hi @mitpre , sorry for my late feedback, the RTC seems working now, thanks