PiSugar/pisugar-power-manager-rs #75: [BUG] PiSugar3; Read soft poweroff flag error warnings

mitpre, 2022-08-16:
**Environment**
 - Pi hardware version: Raspberry Pi Zero W Rev 1.1
 - OS version: Raspbian GNU/Linux 11 (bullseye) (debian 11.4)
 - Application version: 1.7.1
 - PiSugar 3 firmware version: 1.2.6

The `PiSugar3` board is connected to my `pi zero w` and is constantly charging through its USB-C port. It is running more or less normally -- apart from the same RTC problems as described in issue [#74](https://github.com/PiSugar/pisugar-power-manager-rs/issues/74#issue-1328396200). However, I am constantly getting `Read soft poweroff flag error` warning in PiSugar logs (accessible through `journalctl -u pisugar-server`). The examples of warnings are:
1) `Aug 16 07:03:38 rolete pisugar-server[262]: [2022-08-16T05:03:38Z WARN  pisugar_core::pisugar3] Read soft poweroff flag error: I/O error: Remote I/O error (os error 121)`
2) `Aug 16 08:32:51 rolete pisugar-server[262]: [2022-08-16T06:32:51Z WARN  pisugar_core::pisugar3] Read soft poweroff flag error: ctr2 changed during reading`

The first warning is happening multiple times per hour and the second one multiple times per day. So they have a very different frequency.

Can someone explain what is the meaning behind those warnings? What should I do to prevent them?

ctrlaltdude, 2026-03-04:
I had the same problem. It was a bad contact on a pogo pin. Measure the continuety with a multi meter. If a contact is bad, reflow the solder. Then it will work.

JdaieLin, 2026-10-01:
The reported I2C errors are consistent with the pogo-pin contact problem identified in the follow-up. The official FAQ documents the recommended procedure and safety precautions: https://docs.pisugar.com/docs/product-wiki/battery/faq#1-why-pisugar-is-not-able-to-power-raspberry-pi-why-pisugar-23-cannot-be-detected-on-i2c-bus

Power the Raspberry Pi off first, gently clean the underside GPIO solder points with alcohol or a plastic tool, reinstall PiSugar, and only add solder if you are comfortable doing so. Closing as a hardware-contact issue with a documented resolution.

closed 2026-10-01 by JdaieLin