PiSugar/pisugar-power-manager-rs #78: Multiple problems [BUG]

danktankk, 2022-09-12:
**Environment**
 - Pi hardware version: zero w
 - OS version: VERSION="10 (buster)"
 - Application version: 1.7.4

**Additional context**

1:  I can't seem to get the UPS to work as expected.   If I unplug the pi zerow, it reboots instead of the battery taking over to supply power.  If I having it running on battery and plug it in, it also reboots.  Is this normal?  How do you test this if this isn't the way?

2:  The settings in the server don't save after a reboot.  I turn on `battery input protection` and after one of these reboots that spontaneously happens when trying to get the battery to failover in problem number 1 above, the settings is back to default.

3:  I am not getting i2c to work either.  When I run `i2cdetect -y 1`  I get the following:

`pi@PiZeroW-1:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@PiZeroW-1:~ $`

`sudo i2cdump -y 1 0x32` and `sudo i2cdump -y 1 0x75` produce nothing but X's

4:  The RTC clock will not sync and stays at the year 2020.

Any help would be definitely appreciated as I have spent quite a bit of time combing through the "documentation" with no success.

danktankk, 2022-09-13:
no need to reply.  I sent it back and will be buying something other than pisugar,

JdaieLin, 2026-10-02:
The reporter returned the hardware and explicitly requested no further reply, so there is no remaining reproducible case to investigate. Closing this support report as stale.

closed 2026-10-02 by JdaieLin