PiSugar/pisugar-power-manager-rs #92: [BUG] Safe Shutdown not triggering

SHONG0504, 2023-01-19:
**Environment**
 - Pi hardware version: 4b
 - OS version: Pi Desktop
 - Application version: 1.7.4

**To Reproduce**
Steps to reproduce the behavior:
1. Set the soft poweroff level to 30
2. Soft poweroff shell: sudo shutdown -h now
3. Power stays on when battery goes below 30%

**Expected behavior**
Expected to shutdown when battery goes below 30%

**Screenshots**
[image]
[image]

Martyfree123, 2023-02-03:
Having the same issue. Also on a Pi4 with PiSugar 3 Plus

JdaieLin, 2026-10-01:
The automatic-shutdown logic was rewritten and fixed by commit 234e636 and released in v1.7.6. It now tracks the time spent below the configured battery threshold and runs the configured shutdown command when the delay expires. Closing as resolved; please use v1.7.6 or newer.

closed 2026-10-01 by JdaieLin