PiSugar/pisugar-power-manager-rs #100: i2cdump -y 0x75  has ntohing

aidupre, 2023-11-02:
I have a Pi Zero W with a 2 LED Pi Sugar 2 and a v4 waveshare. 
the board is flashed with 1.55 and I'm using the rust version of your power manager.
I have never been able to get it to show things like when its charging. 
I2c is enabled
When I run the i2cdetect -y 1 command I only get a 32 in 30:02
When I run i2cdump -y 1 0x32 I get 00 and 10 filled with information
when I run i2cdump -y 1 0x75 I get nothing.
I have disabled all plugins and run the commands again still nothing in 75
I have completely uninstalled the power manager and reinstalled still nothing in 75.
I am only starting an issue cause the Pwnagotchi locks up a lot when its powered by the pi sugar. If I have if plugged in and the Pisugar off it will run for many hours. last check was 15 without problems. As soon as it gets powered by pisugar it freezes. It has never made it more than 1.5 hrs on Pisugar without rebooting. What could cause this? Thanks

H3CATE, 2023-12-28:
Same thing, received mine today and couldn't find anything online.
running on a pwnagotchi 1.5.5FIX fork and waveshare V4 screen. 
I can see the 0x32 but not 0x75.  The web page doesn't display anything either.
[image]

JdaieLin, 2026-10-01:
Seeing the battery controller at 0x32 but no RTC response at 0x75, together with missing Web UI data, may be caused by poor pogo-pin contact. Please follow the official FAQ procedure here: https://docs.pisugar.com/docs/product-wiki/battery/faq#1-why-pisugar-is-not-able-to-power-raspberry-pi-why-pisugar-23-cannot-be-detected-on-i2c-bus

Power the Raspberry Pi off first, gently clean the underside GPIO solder points with alcohol or a plastic tool, reinstall PiSugar, and test the I2C bus again. Only add solder if you are comfortable doing so. Please report back if 0x75 is still missing afterward.

closed 2026-11-15 by github-actions[bot]