PiSugar/pisugar-power-manager-rs #105: [BUG] Soft Shutdown custom shell script retries too quickly

seelenamt, 2024-02-05:
**Environment**
 - Zero 2W
 - Debian Bookworm
 - 1.7.5

**To Reproduce**
1. Go to the webui > settings > soft shutdown
2. Select custom shell
3. Run a service that takes a few seconds to load, or run any series of commands with a "sleep 10" in there (In my screenshot, longwait.sh is simply "sleep 10 && sudo shutdown now")
4. Press the power button and watch from journalctl as it retries the command about every 0.5 seconds, locking up the device it is running on. (In my screenshot, it never actually reaches the shutdown command, so I believe it is killing the previous attempt every time)
 

**Expected behavior**
The command runs, waiting more than a second or two before killing the running process and retrying it. Ideally, a configurable wait so that the default can be a sane value, but allowing for longer grace periods

**Screenshots**
[image]

mighty-sparrow, 2024-05-03:
I'm having a similar issue wherein I'm trying to clear an ePaper display prior to shutdown. I'm using a python script to manage the ePaper and the command loops over and over, just like was reported by @seelenamt . Is there a GOOD example of how we should be leveraging the soft shutdown script?

gmccauley, 2025-01-18:
I'm having the same issue.  And attempting a similar task as @mighty-sparrow.  I have a bash script that stops a service, clears the display, and then issues the shutdown command.  It probably takes about 15-20'ish seconds to accomplish that when only one process is running.

In my case, it spawns a new process every half a second or so.....  So in 10 seconds, I have more than 20 instances of the same script running.

Any fix for this?  I thought about having the script check to see if another instance was already running and then exit, but that seems like a bit of a bandaid.

mighty-sparrow, 2026-10-03:
Still relevant.