PiSugar/pisugar-power-manager-rs #106: [BUG] PiSugar 2 Pro(Plus) external power on off button

persocon, 2024-02-05:
**Environment**
 - Pi hardware version: Raspberry Pi 4b
 - OS version: Raspbian Bullseye 
 - Application version: PiSugar-server version 1.7.5

**To Reproduce**
Steps to reproduce the behavior:
1. Add button to GPIO 3 to turn on and off
2. Reboot Pi
3. button doesn't work

Alternatively i also have added the shutdown overlay to `/boot/config.txt`
but that one causes conflict:
```
[    5.466851] pinctrl-bcm2835 fe200000.gpio: pin gpio3 already requested by fe804000.i2c; cannot claim for soc:shutdown_button@3
[    5.466886] pinctrl-bcm2835 fe200000.gpio: pin-3 (soc:shutdown_button@3) status -22
[    5.466906] pinctrl-bcm2835 fe200000.gpio: could not request pin 3 (gpio3) from group gpio3  on device pinctrl-bcm2711
[    5.466924] gpio-keys soc:shutdown_button@3: Error applying setting, reverse things back

```

**Expected behavior**
- External button to power on and off
- should i solder one somewhere else?

persocon, 2024-02-20:
meh gave up, support here sucks, removed the board and finding another solution for my project

JdaieLin, 2026-10-02:
GPIO 3 is used by the Raspberry Pi I2C interface, so the gpio-shutdown overlay cannot claim the same pin while I2C is enabled. This is a hardware/pin-assignment conflict rather than a PiSugar Server defect, and the reporter is no longer using the board. Closing as answered/not planned.

closed 2026-10-02 by JdaieLin