PiSugar/pisugar-power-manager-rs #108: [BUG] 404 Error when installing PiSugar 3 software

jockm, 2024-03-22:
**Environment**
 - Pi hardware version: zero
 - OS version: any
 - Application version: don't know

**To Reproduce**
```
wget https://cdn.pisugar.com/release/pisugar-power-manager.sh
bash pisugar-power-manager.sh -c release
```

**Expected behavior**
The software to install

**Console Result**
```
wget https://cdn.pisugar.com/release/pisugar-power-manager.sh
bash pisugar-power-manager.sh -c release
--2024-03-22 09:09:11--  https://cdn.pisugar.com/release/pisugar-power-manager.sh
Resolving cdn.pisugar.com (cdn.pisugar.com)... 8.45.176.227, 8.45.176.231, 8.45.176.230, ...
Connecting to cdn.pisugar.com (cdn.pisugar.com)|8.45.176.227|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4449 (4.3K) [text/plain]
Saving to: ‘pisugar-power-manager.sh’

pisugar-power-manager.sh           100%[=============================================================>]   4.34K  --.-KB/s    in 0s      

2024-03-22 09:09:12 (12.7 MB/s) - ‘pisugar-power-manager.sh’ saved [4449/4449]

Download PiSugar-server and PiSugar-poweroff package 
--2024-03-22 09:09:12--  http://cdn.pisugar.com/release/pisugar-server_1.7.6_armhf.deb
Resolving cdn.pisugar.com (cdn.pisugar.com)... 8.45.176.225, 8.45.176.232, 8.45.176.226, ...
Connecting to cdn.pisugar.com (cdn.pisugar.com)|8.45.176.225|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-03-22 09:09:14 ERROR 404: Not Found.
```

jockm, 2024-03-22:
Also: I get the same result when I choose nightly and not release

Specifically it fails on the line:

`wget -O "$TEMPDIR/${package_server}" "http://cdn.pisugar.com/${channel}/${package_server}"`

which works out to:

`wget -O /tmp/pisugar.8KsEVL/pisugar-server_1.7.6_armhf.deb http://cdn.pisugar.com/release/pisugar-server_1.7.6_armhf.deb`

I have a hard time believing the actual install instructions were actually tested

Alcedema, 2024-03-22:
Have had this issue since 1.7.6 was released. Need this as 1.7.5 frontend can't communicate with the API.

J-CMartin, 2024-03-22:
Came here to report the same problem. Version 1.7.6 doesn't seem to be on CDN.

AnonymousxiosdeV, 2024-03-23:
I have made PR to fix this but for now you can run the following.

`bash pisugar-power-manager.sh -v 1.7.6-1`

JdaieLin, 2026-10-01:
This was a version-specific CDN/package-name mismatch for 1.7.6. Current releases and the current installer no longer request that missing artifact. Closing this obsolete installation issue as resolved.

closed 2026-10-01 by JdaieLin